yoursunny/esp32cam #6: esp32cam::Camera.changeResolution TRUNCATION

alorbach, 2019-10-14:
Firstof all thanks for the esp32cam::Camera.changeResolution feature recently added. 
I am working with it on my own fork which supports ESPAsyncWebServer (Will create PR once this is stable), and I discovered that the memory buffer is not changed according to the Resolution change. 

See debugs from my project: ```
CAPTURE OK 320x240 38400b
CAPTURE OK 640x480 38400b
CAPTURE OK 800x600 38400b
```

So if you start with lower resolution, upper resolutions will have truncated images.

yoursunny, 2019-10-15:
https://github.com/yoursunny/esp32cam/blob/9c4eb5dad6d383a3a6a28dc358aa7a6602de2b92/src/esp32cam.h#L24-L30
As indicated in Doxygen, a precondition of using `changeResolution` function is "initial resolution is higher than specified resolution".
Trying to change resolution to higher than the initial resolution results in undefined behavior.

closed 2019-10-15 by yoursunny

alorbach, 2019-10-16:
I understand no problem, didn't read the comment carefully enough , sorry.