yoursunny/esp32cam #9: Async Request

b2cbd, 2019-12-30:
Great work! Is there any async sample to use GPIO pins while streaming? I can see it uses the default web server library. Is it possible to use it with [asnyc library](https://github.com/me-no-dev/ESPAsyncWebServer)? Any async server sample will be greatly appreciated!

yoursunny, 2019-12-30:
PR #8 has some code related to async server. I'm too busy with other projects that I haven't reviewed it. You can give it a try.

alorbach, 2020-01-15:
@b2cbd: It works very well with my project and ESPAsyncWebServer.

But you will need to start th streaming process in its own Task, otherwise if the Async Callback comes from CORE0, the Watchdog will reset ESP32 within seconds.

RobAxt, 2020-06-28:
Hello **@alorbach** can you give an example? I was just bitten by the dog several times and panicking Core 1 another several times... LOL. Thanks!!

RobAxt, 2020-06-29:
@alorbach I was trying to stream Mjpeg

[main.cpp.txt](https://github.com/yoursunny/esp32cam/files/4843242/main.cpp.txt)

CAMERA OK
IP Address: XXX.xxx.XXX.xxx
STREAM BEGIN
STREAM END 19frm 2.61fps
STREAM BEGIN
STREAM END 45frm 3.01fps
STREAM BEGIN
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40160aaf  PS      : 0x00060b30  A0      : 0x80160cfd  A1      : 0x3ffb1db0
A2      : 0x00000000  A3      : 0x3f401484  A4      : 0x0000006a  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800dd174  A9      : 0x3ffb1ec0
A10     : 0x3ffbedfc  A11     : 0x3f40151f  A12     : 0x3f40151f  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffe5

Backtrace: 0x40160aaf:0x3ffb1db0 0x40160cfa:0x3ffb1df0 0x40160d2b:0x3ffb1e10 0x400d15e9:0x3ffb1e30 0x400d1342:0x3ffb1f60 0x400d152b:0x3ffb1f90 0x400dee41:0x3ffb1fb0 0x4008d871:0x3ffb1fd0

Rebooting...

alorbach, 2020-12-02:
You should install https://github.com/me-no-dev/EspExceptionDecoder to get a readable stack trace., this would help alot to find out where it crashed.

malqattan88, 2021-05-23:
how i add Flashlight (GPIO 4) in the code what should i do?

yoursunny, 2022-02-13:
It's been two years and I finally got around to look at ESPAsyncWebServer library.
Integration is possible but I probably cannot support sending the same image/stream to multiple clients.
So far I added code for capturing a still image.
MJPEG will come in a few days/weeks/months/years.

yoursunny, 2022-02-16:
MJPEG is done too, and I added `AsyncCam` example.

closed 2022-02-16 by yoursunny

beniroquai, 2022-02-27:
Hey, great work! Do you have a feeling about the latency of the `AsyncWebserver` vs. the `Webserver`? Right now it feels very laggy >0.2fps vs. 3-10fps