yoursunny/esp32cam #17: Compilation Issues - resolution.cpp

c3r63ru5, 2020-12-13:
Hey,

I've had issues compiling the example sketches... newest Arduino on MacOS.

Patch below.

Cheers,
Sebastian

```
diff --git a/src/internal/resolution.cpp b/src/internal/resolution.cpp
index 0e9eb38..272bf77 100644
--- a/src/internal/resolution.cpp
+++ b/src/internal/resolution.cpp
@@ -16,7 +16,7 @@ Resolution::getWidth() const
   if (!isValid()) {
     return -1;
   }
-  return ::resolution[m_frameSize][0];
+  return ::resolution[m_frameSize].width;
 }

 int
@@ -25,7 +25,7 @@ Resolution::getHeight() const
   if (!isValid()) {
     return -1;
   }
-  return ::resolution[m_frameSize][1];
+  return ::resolution[m_frameSize].height;
 }

 Resolution
```

yoursunny, 2020-12-13:
This is a breaking change introduced in ESP32 Arduino Core v1.0.5-rc1: structure of `resolutions` element is changed.
https://github.com/espressif/arduino-esp32/commit/22b427df0fc61b44a48769378a105745c4683a81#diff-aa521a4ece89c68dd094eb6682872bdba10f4c69a058292f8c7ca60225816373R96

When v1.0.5 is released, I'll apply the patch.

yoursunny, 2021-02-27:
Applied in e27c0de2d2b2b793fb65100613d1107ae63cf75f.

closed 2021-02-27 by yoursunny