yoursunny/esp32cam #18: Problems with resolution.ccp

Adrianahwjsgsj, 2021-03-02:
Im using your library with platformio but everytime i try to include the library, it returns:

```
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp: In member function 'int esp32cam::Resolution::getWidth() const':
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp:19:36: error: request for member 'width' in 'resolution[((int)((const esp32cam::Resolution*)this)->esp32cam::Resolution::m_frameSize)]', which is of non-class type 'const int [2]'
   return ::resolution[m_frameSize].width;
Compiling .pio\build\esp32cam\lib306\Husarnet\HusarnetClient.cpp.o
                                    ^
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp: In member function 'int esp32cam::Resolution::getHeight() const':
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp:28:36: error: request for member 'height' in 'resolution[((int)((const esp32cam::Resolution*)this)->esp32cam::Resolution::m_frameSize)]', which is 
of non-class type 'const int [2]'
   return ::resolution[m_frameSize].height;
                                    ^
Compiling .pio\build\esp32cam\lib306\Husarnet\HusarnetServer.cpp.o
```

yoursunny, 2021-03-03:
You need to upgrade to ESP32 Arduino Core v1.0.5, see #17.

blackshoals, 2021-12-15:
Unfortunately I'm running into the same error on PlatformIO with platform-espressif32 v3.4.0. Rolling back to 3.1.0 doesn't seem to give me any different result.

`Compiling .pio/build/esp32cam/lib4d6/esp32cam/internal/resolution.cpp.o
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp: In member function 'int esp32cam::Resolution::getWidth() const':
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp:19:36: error: request for member 'width' in 'resolution[((int)((const esp32cam::Resolution*)this)->esp32cam::Resolution::m_frameSize)]', which is of non-class type 'const int [2]'
   return ::resolution[m_frameSize].width;
                                    ^
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp: In member function 'int esp32cam::Resolution::getHeight() const':
.pio/libdeps/esp32cam/esp32cam/src/internal/resolution.cpp:28:36: error: request for member 'height' in 'resolution[((int)((const esp32cam::Resolution*)this)->esp32cam::Resolution::m_frameSize)]', which is of non-class type 'const int [2]'
   return ::resolution[m_frameSize].height;`

yoursunny, 2021-12-17:
This means somehow you still have ESP32 Arduino Core v1.0.4.

blackshoals, 2021-12-21:
Thanks, the Espressif32 platform was up to date and I tried uninstall/reinstall with the same result. PlatformIO clearly makes sense to some users but I found it to be a really confusing universe compared to any other IDE I've tried. Back to the Arduino IDE.

yoursunny, 2022-02-19:
This error is no longer appearing with current PlatformIO.

 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)

But new error https://github.com/yoursunny/esp32cam/issues/23 appears, sigh.

closed 2022-02-19 by yoursunny