yoursunny/esp32cam #23: Getting esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope

al06825, 2022-02-19:
Hello... I upgraded ESP32 to 1.0.6 (latest). Keeps getting this error from your esp32cam.cpp.

\esp32cam-main\src\esp32cam.cpp: In member function 'esp32cam::ResolutionList esp32cam::CameraClass::listResolutions() const':
\esp32cam-main\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
   ^
\esp32cam-main\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                         ^
\esp32cam-main\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

yoursunny, 2022-02-19:
> I upgraded ESP32 to 1.0.6 (latest). 

The latest is [ESP32 Arduino 2.0.2 based on ESP-IDF 4.4-beta1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.2).

mbariola, 2022-02-19:
Hi @yoursunny , I have the same exact problem, but I am using Platformio IDE under VS Code. All libraries are updated, this is my platformio.ini

```

[platformio]
default_envs = esp32cam
build_dir = ${sysenv.TEMP}/pio-build/$PROJECT_HASH

[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
monitor_speed = 460800
upload_speed = 460800
monitor_filters = esp32_exception_decoder, send_on_enter, colorize, time
; change MCU frequency
;board_build.f_cpu = 80000000L
lib_deps =
    yoursunny/esp32cam@^0.0.20220215
```

These are the compile errors I get
```
Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (3.5.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <esp32cam> 0.0.20220215
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling C:\users\massi\AppData\Local\Temp\pio-build\ESP32CAM_OPENCV-6a1a552529\esp32cam\lib578\esp32cam\esp32cam.cpp.o
.pio\libdeps\esp32cam\esp32cam\src\esp32cam.cpp: In member function 'esp32cam::ResolutionList esp32cam::CameraClass::listResolutions() const':
.pio\libdeps\esp32cam\esp32cam\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
   ^
.pio\libdeps\esp32cam\esp32cam\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                         ^
.pio\libdeps\esp32cam\esp32cam\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                                                                      ^
*** [C:\users\massi\AppData\Local\Temp\pio-build\ESP32CAM_OPENCV-6a1a552529\esp32cam\lib578\esp32cam\esp32cam.cpp.o] Error 1
====================================================== [FAILED] Took 1.53 seconds ======================================================
The terminal process "C:\Users\Massi\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

I know that platformio usually has issues with functions and entities used before they are declared, but this does not seem to be the case. Am I missing anything that I should include or could you suggest where to look? Thanks.

yoursunny, 2022-02-19:
> ` - framework-arduinoespressif32 3.10006.210326 (1.0.6)`

You are using arduino-esp32 1.0.6.
It needs to be the latest version 2.0.2.

See also https://github.com/platformio/platform-espressif32/issues/619

mbariola, 2022-02-19:
Ouch. How did I manage to miss that. 
I'll try the `upstream` branch, and if it doesn't work, I'll try the previous version of the lib. Thanks so much for the patience @yoursunny

yoursunny, 2022-02-19:
The following seems to work with `WifiCam` example.

**platformio.ini** content:

```ini
[env:esp32cam]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32cam
framework = arduino

platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2

lib_deps =
  yoursunny/esp32cam @ 0.0.20220215
```

Build output:

```text
Processing esp32cam (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32cam; framework: arduino)
---------------------------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.caef400 
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 8.4.0+2021r2
Converting WifiCam.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <esp32cam> 0.0.20220215
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
Building in release mode
Compiling .pio\build\esp32cam\src\WifiCam.ino.cpp.o
Compiling .pio\build\esp32cam\src\handlers.cpp.o
Generating partitions .pio\build\esp32cam\partitions.bin
Compiling .pio\build\esp32cam\lib04a\esp32cam\esp32cam.cpp.o
Compiling .pio\build\esp32cam\lib04a\esp32cam\internal\config.cpp.o
Compiling .pio\build\esp32cam\lib04a\esp32cam\internal\frame.cpp.o
Compiling .pio\build\esp32cam\lib04a\esp32cam\internal\mjpeg.cpp.o
Compiling .pio\build\esp32cam\lib04a\esp32cam\internal\resolution.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiClient.cpp.o
Archiving .pio\build\esp32cam\lib04a\libesp32cam.a
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32cam\libcaf\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32cam\libbce\FS\FS.cpp.o
Compiling .pio\build\esp32cam\libbce\FS\vfs_api.cpp.o
Compiling .pio\build\esp32cam\lib651\WebServer\Parsing.cpp.o
[... 37 lines cut ...]
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32cam\libFrameworkArduino.a
Linking .pio\build\esp32cam\firmware.elf
Retrieving maximum program size .pio\build\esp32cam\firmware.elf
Checking size .pio\build\esp32cam\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.9% (used 45412 bytes from 327680 bytes)
Flash: [======    ]  60.1% (used 788177 bytes from 1310720 bytes)
Building .pio\build\esp32cam\firmware.bin
esptool.py v3.1
Merged 25 ELF sections
===================================== [SUCCESS] Took 59.01 seconds =====================================
```

mbariola, 2022-02-19:
I confirm that the `upstream` workaround allows for compilation. thanks again

al06825, 2022-02-20:
Wow TQ all for quick replies. I will give the above all a try. Will update later.

closed 2022-02-20 by yoursunny

anjalir14, 2022-02-21:
> Wow TQ all for quick replies. I will give the above all a try. Will update later.
> 
> HOURS LATER.... The above error is gone with the upgrade of ESP32 2.0 firmware. Really appreciate the advice.

Hlo,

Can you please let me know how to upgrade ESP32 2.0 . Since I tried many ways but still I'm getting the same error.

yoursunny, 2022-02-21:
> Can you please let me know how to upgrade ESP32 2.0 .

See https://github.com/yoursunny/esp32cam/issues/23#issuecomment-1046120261

anjalir14, 2022-02-22:
> > Can you please let me know how to upgrade ESP32 2.0 .
> 
> See [#23 (comment)](https://github.com/yoursunny/esp32cam/issues/23#issuecomment-1046120261)

Thanks for your patience and quick reply. 
I already tried this way but I'm still getting the same error.

skeerthan98, 2022-02-22:
Can you please let me know how to upgrade to v2.0.2 on Arduino IDE?

yoursunny, 2022-02-22:
> Can you please let me know how to upgrade ESP32 2.0 .
> I'm using Arduino IDE

In Arduino IDE,

* if you installed via Boards Manager, you can upgrade/downgrade arduino-esp32 in Tools-Board-Boards Manager menu. There's a dropdown menu letting you choose version.
* if you installed arduino-esp32via git, you can run `git pull`.

See also https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/installing.html

anjalir14, 2022-02-23:
> In Arduino IDE,
>
> * if you installed via Boards Manager, you can upgrade/downgrade arduino-esp32 in Tools-Board-Boards Manager menu. There's a dropdown menu letting you choose version.
> * if you [...]

Thanks a lot. That error is solved but now I'm getting another error in that. Please help me.
The error is:
C:\Users\DNK\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp: In member function 'bool WiFiGenericClass::setDualAntennaConfig(uint8_t, uint8_t, wifi_rx_ant_t, wifi_tx_ant_t)':
C:\Users\DNK\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1236:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\Users\DNK\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1236:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\DNK\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1236:5: sorry, unimplemented: non-trivial designated initializers not supported

Pls help me

yoursunny, 2022-02-23:
@anjalir14 
> C:\Users\DNK\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1236:5: sorry, unimplemented: non-trivial designated initializers not supported

This is not from my library.
It's recommended to install ESP32 Arduino core via Boards Manager and delete the git clone.
If you still cannot solve, ask in [ESP32 Forum](https://www.esp32.com/).

anjalir14, 2022-02-23:
> @anjalir14
>
>
> This is not from my library. It's recommended to install ESP32 Arduino core via Boards Manager and delete the git clone. If you still cannot solve, ask in [ESP32 [...]

Thanks for your reply. I solved my error and now it started working properly.

InvinsibleFox, 2022-03-28:
I keep having this problem and i can find out how to solve it. I really need help.

Processing esp32cam (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32cam; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2   
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2  
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2   
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2  
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.caef400
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 8.4.0+2021r2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <esp32cam> 0.0.20220215
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
Building in release mode
Compiling .pio\build\esp32cam\src\main.cpp.o
Compiling .pio\build\esp32cam\lib792\esp32cam\esp32cam.cpp.o
Compiling .pio\build\esp32cam\lib792\esp32cam\internal\frame.cpp.o
Compiling .pio\build\esp32cam\lib792\esp32cam\internal\mjpeg.cpp.o
Compiling .pio\build\esp32cam\libfff\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32cam\libfff\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32cam\libfff\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32cam\libfff\WiFi\WiFiGeneric.cpp.o
In file included from C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/libraries/WiFi/src/WiFiClient.h:24,
                 from C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/libraries/WiFi/src/WiFi.h:37,
                 from C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/libraries/WebServer/src/WebServer.h:29,      
                 from src/main.cpp:1:
C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/cores/esp32/Arduino.h:201:10: fatal error: pins_arduino.h: No such 
file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32cam\src\main.cpp.o] Error 1
In file included from .pio/libdeps/esp32cam/esp32cam/src/internal/frame.cpp:4:
C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/cores/esp32/Arduino.h:201:10: fatal error: pins_arduino.h: No such 
file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
[... 80 lines cut ...]
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32cam\libfff\WiFi\WiFiAP.cpp.o] Error 1
In file included from C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/libraries/WiFi/src/WiFiClient.h:24,
                 from C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/libraries/WiFi/src/WiFi.h:37,
                 from C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/libraries/WiFi/src/WiFiGeneric.cpp:25:       
C:/Users/Taka/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/cores/esp32/Arduino.h:201:10: fatal error: pins_arduino.h: No such 
file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32cam\libfff\WiFi\WiFiGeneric.cpp.o] Error 1
====================================================================== [FAILED] Took 7.22 seconds ======================================================================

InvinsibleFox, 2022-03-28:
Thank you so much

mohcly, 2022-04-01:
> Wow TQ all for quick replies. I will give the above all a try. Will update later.
> 
> HOURS LATER.... The above error is gone with the upgrade of ESP32 2.0 firmware. Really appreciate the advice.

hello, please how have you done the update??

SirMegaMU, 2022-05-14:
sorry but I have updated my arduino esp32 pack, but it still dosen't work
~~~
Arduino:1.8.19 (Windows 10), 开发板："AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

D:\Documents\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp: In member function 'esp32cam::ResolutionList esp32cam::CameraClass::listResolutions() const':

D:\Documents\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope

   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

   ^

D:\Documents\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope

   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

                         ^

D:\Documents\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope

   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

                                                                      ^

exit status 1

为开发板 AI Thinker ESP32-CAM 编译时出错。
~~~