yoursunny/esp32cam #25: camera_sensor_info_t not declared

wuppergit, 2022-03-03:
Example WiFi-CAM (and others) dont't work. Reason compiler-error:
C:\Users\c\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

and next:
C:\Users\c\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                         ^
C:\Users\c\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

Any idea

yoursunny, 2022-03-03:
Duplicate of #23.

wuppergit, 2022-03-03:
thank you so much & sorry

closed 2022-03-03 by yoursunny