yoursunny/esp32cam #29: error: this site cannot be reached

reemastha706, 2022-04-29:
I am using the library, however, even though the camera is initialized and the web server IP address is shown in the serial monitor, I cannot access the site.  I haven't used the IP address anywhere else.

reemastha706, 2022-04-29:
> 1. Install Wireshark on your computer.
> 2. Select the network interface that you use to connect to the camera. If you have both wired Ethernet and WiFi, consider disabling one so that you can be certain which interface is being used.
> 3. Type `host 192.0.2.1` (substitute the IP address with what's shown on the serial console) in the Wireshark filter box, start capturing.
> 4. After you attempt to access the camera, stop capturing in Wireshark, and save as .pcap or .pcapng file.
> 5. Upload the capture file.

Hi. Thanks for your reply. I thought that I messed up something while downloading this library, however, it seems that I have been doing something wrong as even with the example code from esp32cam, the webserver couldn't be reached. It was working fine before. I want to use the video stream in opencv so Wireshark might not be the good option, right?

yoursunny, 2022-04-29:
> I want to use the video stream in opencv so Wireshark might not be the good option, right?

You need to use either Wireshark or `tcpdump` to determine why your computer cannot reach the camera.

reemastha706, 2022-04-29:
Thanks. Problem was with my wifi router. When I tried hotspot from ethernet, it worked.

closed 2022-04-29 by reemastha706