yoursunny/esp32cam #32: Wificam program error

MunyitLearning, 2022-06-10:
excuse me, i have problem related to library in link (https://github.com/yoursunny/esp32cam/commit/e27c0de2d2b2b793fb65100613d1107ae63cf75f
)
error =
C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-e27c0de2d2b2b793fb65100613d1107ae63cf75f\src\internal\resolution.cpp: In member function 'int esp32cam::Resolution::getWidth() const':

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-e27c0de2d2b2b793fb65100613d1107ae63cf75f\src\internal\resolution.cpp:19:36: error: request for member 'width' in 'camresolution[((int)((const esp)32 ::Resolution*)this)->esp32cam::Resolution::m_frameSize)]', which is of non-class type 'const int [2]'

   return ::resolution[m_frameSize].width;

                                    ^

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-e27c0de2d2b2b793fb65100613d1107ae63cf75f\src\internal\resolution.cpp: In member function 'int esp32cam::Resolution::getHeight() const':

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-e27c0de2d2b2b793fb65100613d1107ae63cf75f\src\internal\resolution.cpp:28:36: error: request for member 'height' in 'resolution[((int)((const esp)((const esp) ::Resolution*)this)->esp32cam::Resolution::m_frameSize)]', which is of non-class type 'const int [2]'

   return ::resolution[m_frameSize].height;

MunyitLearning, 2022-06-10:
resolution.cpp problem, how to solve it?

yoursunny, 2022-06-10:
Upgrade both arduino-esp32 and esp32cam to the latest version.

MunyitLearning, 2022-06-10:
please, share the link esp32cam sir. Thank you in advance

MunyitLearning, 2022-06-10:
ok sir i have upgraded to the latest version. (https://github.com/yoursunny/esp32cam)

error =
ResolveLibrary(uri/UriBraces.h)handlers.cpp:3:27: fatal error: uri/UriBraces.h: No such file or directory

   -> candidates: []
compilation terminated.

Using library esp32cam-e27c0de2d2b2b793fb65100613d1107ae63cf75f at version 0.0.20220215 in folder: C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-e27c0de2d2b2b793fbae6563cf753df11073
Using library WebServer at version 1.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer
Using library WiFi at version 1.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
exit status 1
uri/UriBraces.h: No such file or directory

library #include <uri/UriBraces.h> can't be read what is the solution?

yoursunny, 2022-06-10:
Latest esp32cam is b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9 ; you have e27c0de2d2b2b793fb65100613d1107ae63cf75f.
Latest arduino-esp32 is 2.0.3 ; you have 1.0.4.

Since my library hooks into arduino-esp32 internals, it's quite picky on versions.

MunyitLearning, 2022-06-10:
done (Latest esp32cam is b20ffa9 ; you have e27c0de.)
sorry sir, (Latest arduino-esp32 is 2.0.3 ; you have 1.0.4.) is this for library or board?
I've upgraded the board in Arduino IDE only up to 1.0.6

and error

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9\src\esp32cam.cpp: In member function 'esp32cam::ResolutionList esp32cam::'CasomeraClass:()

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope

   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

   ^

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope

   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

                         ^

C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope

   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

                                                                      ^

Using library esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9 at version 0.0.20220215 in folder: C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-b20ffa9ad2052e22355d4d6f5091c6fd4b70961c6d2052
Using library WebServer at version 1.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer
Using library WiFi at version 1.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
Using library FS at version 1.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS
exit status 1
Compiling error for the ESP32 Wrover Module board.

MunyitLearning, 2022-06-10:
sorry sir, library or board?
I have upgraded the board in Arduino IDE only up to 1.0.6

yoursunny, 2022-06-11:
Upgrade the board to 2.0.3.

MunyitLearning, 2022-06-11:
ok done upgrade board v.2.0.6

and error again

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp: In static member function 'static bool WiFiGenericClass::setDualAntennaConfig(uint8_t, uint8_t, wifi_rx_ant_t, wifi_tx_ant_t)':

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported

     };

     ^

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported

"C:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.1-1-gb8050b365e\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/config" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/heap/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/log/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/soc/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/soc/esp32" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/hal/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_common/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_system/include" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Users\\ASUS PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Users\\ASUS PC\\AppData [... cut ...]
Using library esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9 at version 0.0.20220215 in folder: C:\Users\ASUS PC\Documents\Arduino\libraries\esp32cam-b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9 
Using library WebServer at version 2.0.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WebServer 
Using library WiFi at version 2.0.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi 
Using library FS at version 2.0.0 in folder: C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\FS 
exit status 1
Error compiling for board ESP32 Wrover Module.

MunyitLearning, 2022-06-11:
error like this sir

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp: In static member function 'static bool WiFiGenericClass::setDualAntennaConfig(uint8_t, uint8_t, wifi_rx_ant_t, wifi_tx_ant_t)':

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported

     };

     ^

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported

C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported

yoursunny, 2022-06-11:
> C:\Users\ASUS PC\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++

The compiler that comes with arduino-esp32 2.0.3 would be [gcc8_4_0-esp-2021r2-patch3](https://github.com/espressif/arduino-esp32/blob/142fceb8563cd1795d619829e0a103770a344e1a/package/package_esp32_index.template.json#L57).
You didn't install the board correctly.

MunyitLearning, 2022-06-11:
how to reinstall sir? any tutorials?

MunyitLearning, 2022-06-12:
"done compiling" in preferences must include (https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json)
previously included (https://dl.espressif.com/dl/package_esp32_index.json) so only last update 1.0.6
thank you very much sir, God bless you

closed 2022-06-12 by yoursunny