yoursunny/esp32cam #33: Brownout detector doubt

beckmx, 2022-06-11:
Hello, I was testing the examples, but no matter what example I use, they go to brownout on the WIFI, I tested on several boards and several programmers to provide enough voltage, but still getting that, is is because of a configuration in power inside the library? because the normal esp_who does work for me.

Greetings!

yoursunny, 2022-06-11:
I also see brownout reset frequently, and I don't know why.

beckmx, 2022-06-11:
I just solved it, the wifi config needs to come first before enabling the camera, it makes total sense that the wifi uses so much power and doesnt make a lot of sense to start the camera if the wifi is off, so just by inverting the init stage, the brown out can be avoided.

closed 2022-06-11 by beckmx