yoursunny/esp32cam #34: Very low FPS when trying the AsyncCam example

chewbaz1, 2022-06-26:
Hello

I tried using the AsyncCam example but I get very poor fps (1-2fps). When i use the standard example from espressif I get normal fps. Changing resolution doesnt help. I have an AI Thinker board.  I am using PlatformIO and have tried upload with both AI thinker setting and Wrover setting but fps is bad on both. When I tried using arduino IDE i got a lot of errors therefore I use platformio instead. 

I would be happy if I recieved help with this issue.

Thanks

ery346, 2022-10-09:
It is a very good library but I am also having the same problem

Matt-Stedman, 2024-01-01:
Same here. I can get a v. good frame rate if I'm on StreamMjpeg, but the async example is depressingly slow!

yoursunny, 2024-01-13:
> When i use the standard example from espressif I get normal fps.

Do they offer an example with AsyncTCP?
If so, can you post a link?

bsdshneg, 2024-03-20:
+1
very slow video ~1fps

yoursunny, 2024-06-18:
I added some debug logging in `MjpegResponse` and can have some insights in where it's spending the most time.
In my location the 2.4GHz network is very noisy, such that the home router indicates "RX rate 6 Mbit/s" from the ESP32.
I tested with 640x480 resolution for 30 seconds and got 1.393932 fps.

<details>
<summary>full serial logs</summary>
```text
[   32941] MjpegResponse(0x3ffdb8ec) created
[   32942] MjpegResponse(0x3ffdb8ec) capturing
[   33417] MjpegResponse(0x3ffdb8ec) frame has 7352 octets
[   33532] MjpegResponse(0x3ffdb8ec) sent to client
[   33917] MjpegResponse(0x3ffdb8ec) capturing
[   34417] MjpegResponse(0x3ffdb8ec) frame has 7357 octets
[   34513] MjpegResponse(0x3ffdb8ec) sent to client
[   34517] MjpegResponse(0x3ffdb8ec) capturing
[   34917] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   34999] MjpegResponse(0x3ffdb8ec) sent to client
[   35005] MjpegResponse(0x3ffdb8ec) capturing
[   35417] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   35701] MjpegResponse(0x3ffdb8ec) sent to client
[   35706] MjpegResponse(0x3ffdb8ec) capturing
[   35917] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   36010] MjpegResponse(0x3ffdb8ec) sent to client
[   36417] MjpegResponse(0x3ffdb8ec) capturing
[   36917] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   37266] MjpegResponse(0x3ffdb8ec) sent to client
[   37269] MjpegResponse(0x3ffdb8ec) capturing
[   37417] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   37505] MjpegResponse(0x3ffdb8ec) sent to client
[   37514] MjpegResponse(0x3ffdb8ec) capturing
[   37917] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   38361] MjpegResponse(0x3ffdb8ec) sent to client
[   38417] MjpegResponse(0x3ffdb8ec) capturing
[   38917] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   39029] MjpegResponse(0x3ffdb8ec) sent to client
[   39033] MjpegResponse(0x3ffdb8ec) capturing
[   39417] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   39514] MjpegResponse(0x3ffdb8ec) sent to client
[   39514] MjpegResponse(0x3ffdb8ec) sent to client
[   39917] MjpegResponse(0x3ffdb8ec) capturing
[   40417] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   40513] MjpegResponse(0x3ffdb8ec) sent to client
[   40519] MjpegResponse(0x3ffdb8ec) capturing
[   40917] MjpegResponse(0x3ffdb8ec) frame has 7148 octets
[   41058] MjpegResponse(0x3ffdb8ec) sent to client
[   41417] MjpegResponse(0x3ffdb8ec) capturing
[   41917] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   42172] MjpegResponse(0x3ffdb8ec) sent to client
[   42173] MjpegResponse(0x3ffdb8ec) capturing
[   42417] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   42501] MjpegResponse(0x3ffdb8ec) sent to client
[   42917] MjpegResponse(0x3ffdb8ec) capturing
[   43417] MjpegResponse(0x3ffdb8ec) frame has 7166 octets
[   43754] MjpegResponse(0x3ffdb8ec) sent to client
[   43918] MjpegResponse(0x3ffdb8ec) capturing
[   43921] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   44006] MjpegResponse(0x3ffdb8ec) sent to client
[   44013] MjpegResponse(0x3ffdb8ec) capturing
[   44418] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   44562] MjpegResponse(0x3ffdb8ec) sent to client
[   44563] MjpegResponse(0x3ffdb8ec) capturing
[   44904] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   44947] MjpegResponse(0x3ffdb8ec) sent to client
[   45002] MjpegResponse(0x3ffdb8ec) capturing
[   45418] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   45520] MjpegResponse(0x3ffdb8ec) sent to client
[   45522] MjpegResponse(0x3ffdb8ec) capturing
[   45918] MjpegResponse(0x3ffdb8ec) frame has 7356 octets
[   46014] MjpegResponse(0x3ffdb8ec) sent to client
[   46020] MjpegResponse(0x3ffdb8ec) capturing
[... 60 lines cut ...]
[   61418] MjpegResponse(0x3ffdb8ec) frame has 7164 octets
[   61580] MjpegResponse(0x3ffdb8ec) sent to client
[   61628] MjpegResponse(0x3ffdb8ec) capturing
[   61918] MjpegResponse(0x3ffdb8ec) frame has 7164 octets
[   62021] MjpegResponse(0x3ffdb8ec) sent to client
[   62418] MjpegResponse(0x3ffdb8ec) capturing
[   62918] MjpegResponse(0x3ffdb8ec) frame has 7161 octets
[   63058] MjpegResponse(0x3ffdb8ec) sent to client
[   63130] MjpegResponse(0x3ffdb8ec) capturing
[   63418] MjpegResponse(0x3ffdb8ec) frame has 7152 octets
[   63789] MjpegResponse(0x3ffdb8ec) deleted after 43 frames at fps 1.393932
```
</details>

Some spreadsheet calculation reveals the average wait time:

1. Waiting to start capture: 133ms
2. Camera performing the capture: 224ms
3. Sending the frame to the client: 336ms

In the current codebase, the **MjpegController** type can only make use of one `Frame`, even if multiple frames are created via `Config::setBufferCount`.
If this limitation is resolved, the duration for "camera performing the capture" and "sending the frame to the client" can be overlapped, which would greatly improve the fps.

Separately, it should be possible to refactor **MjpegResponse** type to reduce or eliminate the "waiting to start capture" duration.
While this is a smaller improvement, it is much easier to achieve.

yoursunny, 2024-06-18:
I tried to reduce the SEND-CAPTURE wait in 00d1ddbd2a0c93201364c03320674c62fbbaa1d5.

<details><summary>full serial logs</summary>

```text
[  152698] MjpegResponse(0x3ffdc148) created
[  152698] MjpegResponse(0x3ffdc148) capturing
[  153189] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  154022] MjpegResponse(0x3ffdc148) sent to client
[  154023] MjpegResponse(0x3ffdc148) capturing
[  154032] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  154640] MjpegResponse(0x3ffdc148) sent to client
[  154640] MjpegResponse(0x3ffdc148) capturing
[  154689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  155811] MjpegResponse(0x3ffdc148) sent to client
[  155811] MjpegResponse(0x3ffdc148) capturing
[  156142] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  157066] MjpegResponse(0x3ffdc148) sent to client
[  157066] MjpegResponse(0x3ffdc148) capturing
[  157142] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  158189] MjpegResponse(0x3ffdc148) sent to client
[  158189] MjpegResponse(0x3ffdc148) capturing
[  158689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  159553] MjpegResponse(0x3ffdc148) sent to client
[  159553] MjpegResponse(0x3ffdc148) capturing
[  159689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  161121] MjpegResponse(0x3ffdc148) sent to client
[  161121] MjpegResponse(0x3ffdc148) capturing
[  161189] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  161959] MjpegResponse(0x3ffdc148) sent to client
[  161959] MjpegResponse(0x3ffdc148) capturing
[  162146] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  162577] MjpegResponse(0x3ffdc148) sent to client
[  162577] MjpegResponse(0x3ffdc148) capturing
[  162689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  163189] MjpegResponse(0x3ffdc148) sent to client
[  163189] MjpegResponse(0x3ffdc148) capturing
[  163237] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  164896] MjpegResponse(0x3ffdc148) sent to client
[  164897] MjpegResponse(0x3ffdc148) capturing
[  165189] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  165713] MjpegResponse(0x3ffdc148) sent to client
[  165713] MjpegResponse(0x3ffdc148) capturing
[  165743] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  166330] MjpegResponse(0x3ffdc148) sent to client
[  166330] MjpegResponse(0x3ffdc148) capturing
[  166338] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  167039] MjpegResponse(0x3ffdc148) sent to client
[  167039] MjpegResponse(0x3ffdc148) capturing
[  167162] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  168189] MjpegResponse(0x3ffdc148) sent to client
[  168189] MjpegResponse(0x3ffdc148) capturing
[  168689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  170725] MjpegResponse(0x3ffdc148) sent to client
[  170725] MjpegResponse(0x3ffdc148) capturing
[  170776] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  171578] MjpegResponse(0x3ffdc148) sent to client
[  171578] MjpegResponse(0x3ffdc148) capturing
[  171689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  172427] MjpegResponse(0x3ffdc148) sent to client
[  172427] MjpegResponse(0x3ffdc148) capturing
[  172514] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  173189] MjpegResponse(0x3ffdc148) sent to client
[  173189] MjpegResponse(0x3ffdc148) capturing
[  173196] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  173680] MjpegResponse(0x3ffdc148) sent to client
[  173680] MjpegResponse(0x3ffdc148) capturing
[  173689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  174072] MjpegResponse(0x3ffdc148) sent to client
[  174072] MjpegResponse(0x3ffdc148) capturing
[  174189] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  174563] MjpegResponse(0x3ffdc148) sent to client
[  174563] MjpegResponse(0x3ffdc148) capturing
[  174689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  175366] MjpegResponse(0x3ffdc148) sent to client
[  175366] MjpegResponse(0x3ffdc148) capturing
[  175479] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  176189] MjpegResponse(0x3ffdc148) sent to client
[  176189] MjpegResponse(0x3ffdc148) capturing
[  176419] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  177189] MjpegResponse(0x3ffdc148) sent to client
[  177189] MjpegResponse(0x3ffdc148) capturing
[  177271] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  177594] MjpegResponse(0x3ffdc148) sent to client
[  177594] MjpegResponse(0x3ffdc148) capturing
[  177689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  178856] MjpegResponse(0x3ffdc148) sent to client
[  178856] MjpegResponse(0x3ffdc148) capturing
[  179163] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  180030] MjpegResponse(0x3ffdc148) sent to client
[  180030] MjpegResponse(0x3ffdc148) capturing
[  180186] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  181577] MjpegResponse(0x3ffdc148) sent to client
[  181577] MjpegResponse(0x3ffdc148) capturing
[  181689] MjpegResponse(0x3ffdc148) frame has 7922 octets
[  182973] MjpegResponse(0x3ffdc148) deleted after 29 frames at fps 0.957886
```
</details>

According to the spreadsheet, each step takes average time as follows:

1. Waiting to start capture: 0ms
2. Camera performing the capture: 140ms
3. Sending the frame to the client: 857ms

The change is effective in eliminating the SEND-CAPTURE delay, but the duration spent within SEND state is much longer.
I can't tell whether it's due to WiFi interference or something in the code.

yoursunny, 2025-01-09:
I introduced a *continuous mode* in commit d23dbfe96bf4e8b0dbbe107f9aae56f373d75f3e.
The main difference is that, normally MjpegResponse starts a `capture()` after the previous frame is fully sent, while in continuous mode it can start a `capture()` (into a separate frame buffer) while the previous frame is still being sent.

In the AsyncCam example, `/cam.mjpeg` is the normal mode, `/continuous.mjpeg` is the continuous mode.
I tested with 640x480 resolution for 300 seconds.

<details><summary>normal mode 1.923 fps</summary>

```text
[  326767] MjpegController(0x3ffdfdcc) notifyCapture next=326767
[  327094] MjpegController(0x3ffdfdcc) notifyReturn frame=0x3f84d780 size=8182 dimension=640x480
[  327183] MjpegController(0x3ffdfdcc) notifySent count=573 ok=1
[  327183] MjpegController(0x3ffdfdcc) notifyCapture next=327183
[  327594] MjpegController(0x3ffdfdcc) notifyReturn frame=0x3f800a60 size=8182 dimension=640x480
[  327697] MjpegController(0x3ffdfdcc) notifySent count=574 ok=1
[  327697] MjpegController(0x3ffdfdcc) notifyCapture next=327697
[  328094] MjpegController(0x3ffdfdcc) notifyReturn frame=0x3f8270f0 size=8182 dimension=640x480
[  328214] MjpegController(0x3ffdfdcc) notifySent count=575 ok=1
[  328214] MjpegController(0x3ffdfdcc) notifyCapture next=328214
[  328594] MjpegController(0x3ffdfdcc) notifyReturn frame=0x3f84d780 size=8182 dimension=640x480
[  328735] MjpegController(0x3ffdfdcc) notifySent count=576 ok=1
[  328735] MjpegController(0x3ffdfdcc) notifyCapture next=328735
[  329094] MjpegController(0x3ffdfdcc) notifyReturn frame=0x3f800a60 size=8182 dimension=640x480
[  329333] MjpegController(0x3ffdfdcc) notifySent count=577 ok=1
[  329334] MjpegController(0x3ffdfdcc) notifyCapture next=329334
[  329594] MjpegController(0x3ffdfdcc) notifyReturn frame=0x3f8270f0 size=8182 dimension=640x480
[  329616] MjpegResponse(0x3ffdfd4c) deleted after 577 frames at fps 1.923199

$ curl -o /dev/null -m 300 http://192.168.137.163/cam.mjpeg
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 4662k    0 4662k    0     0  15913      0 --:--:--  0:05:00 --:--:-- 16836
curl: (28) Operation timed out after 300003 milliseconds with 4774098 bytes received
```

</details>
<details><summary>continuous mode 2.033 fps</summary>

```text
[  305852] MjpegController(0x3ffe066c) notifyCapture next=305852
[  305884] MjpegController(0x3ffe066c) notifyReturn frame=0x3f84d780 size=8701 dimension=640x480
[  306275] MjpegController(0x3ffe066c) notifySent count=606 ok=1
[  306275] MjpegController(0x3ffe066c) notifyCapture next=306275
[  306384] MjpegController(0x3ffe066c) notifyReturn frame=0x3f800a60 size=9457 dimension=640x480
[  306569] MjpegController(0x3ffe066c) notifySent count=607 ok=1
[  306569] MjpegController(0x3ffe066c) notifyCapture next=306569
[  306884] MjpegController(0x3ffe066c) notifyReturn frame=0x3f8270f0 size=8607 dimension=640x480
[  307109] MjpegController(0x3ffe066c) notifySent count=608 ok=1
[  307109] MjpegController(0x3ffe066c) notifyCapture next=307109
[  307384] MjpegController(0x3ffe066c) notifyReturn frame=0x3f84d780 size=9994 dimension=640x480
[  307516] MjpegController(0x3ffe066c) notifySent count=609 ok=1
[  307516] MjpegController(0x3ffe066c) notifyCapture next=307516
[  307598] MjpegController(0x3ffe066c) notifyReturn frame=0x3f800a60 size=9180 dimension=640x480
[  307798] MjpegController(0x3ffe066c) notifySent count=610 ok=1
[  307799] MjpegController(0x3ffe066c) notifyCapture next=307799
[  307884] MjpegController(0x3ffe066c) notifyReturn frame=0x3f8270f0 size=8597 dimension=640x480
[  308001] MjpegResponse(0x3ffe05ec) deleted after 610 frames at fps 2.033916

$ curl -o /dev/null -m 300 http://192.168.137.163/continuous.mjpeg
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 5540k    0 5540k    0     0  18911      0 --:--:--  0:05:00 --:--:-- 22934
curl: (28) Operation timed out after 300009 milliseconds with 5673702 bytes received
```

</details>