yoursunny/esp32cam #35: Capturing problem

orkungedik, 2022-06-30:
Hi,

My system using YYH-v2.0 camera running on ESP32 CAM board. I uploaded sample code, and face with "capture error" when I try to capture even 96x96 JPEG. What may cause the problem?

Board -> ESP32 Wrover Module
Partition Scheme -> Default 4MB with spiffs

Thank you

Orkun Gedik

yoursunny, 2022-07-19:
The hardware may not support every resolution.
Try a different resolution.

greenrat, 2022-09-04:
Hi orkungedik, I have the same camera YYH-v2.0 with a module ESP32 CAM. My code is being filled in, but the signal from the camera is broken. I searched for information on this camera and found nothing. Did you manage to find something about this camera ?
[image]

orkungedik, 2022-09-05:
Hi GambitOZ,

Your problem may be fixed by one of the possible two reasons, below;

1)  It never display color by my side. But I was able to display in gray scale. Then I bought a brand new OV2640 camera instead of using YYH-v2.0. If you don't decide to change camera  device, change pixel format by PIXFORMAT_GRAYSCALE in your camera_config_t structure. This trick worked on my system.

`config.pixel_format = PIXFORMAT_JPEG;` to `config.pixel_format = PIXFORMAT_GRAYSCALE;`

2) You may face with brownout detector. So you can check messages your serial output. If it is brownout problem, I suggest you provide dedicated power 5V+ source  to you Camera device

Best regards,

Orkun Gedik

GambitOZ, 2022-09-05:
Thanks for the help, Everything is fine with the power supply, I have a powerful 3.3 volt power supply unit.
With the transition to gray, the picture began to be displayed. And it slows down terribly 5 second. I'll try to experiment with the code, if something interesting turns out, I'll write a solution to the problem here.