yoursunny/esp32cam #36: cannot recognize board

mohammadbara, 2022-07-23:
hi.

exit status 1
Error compiling for board ESP32 Wrover Module.

yoursunny, 2022-07-23:
What's the complete error message?

mohammadbara, 2022-07-24:
C:\Users\Mohammad\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp: In member function 'esp32cam::ResolutionList esp32cam::CameraClass::listResolutions() const':
C:\Users\Mohammad\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
   ^
C:\Users\Mohammad\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                         ^
C:\Users\Mohammad\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                                                                      ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Mohammad\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: F:\Soft Ware\arduino\arduino-1.8.19\libraries\WiFi
Using library WebServer at version 1.0 in folder: C:\Users\Mohammad\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer 
Using library WiFi at version 1.0 in folder: C:\Users\Mohammad\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi 
Using library esp32cam-main at version 0.0.20220612 in folder: C:\Users\Mohammad\Documents\Arduino\libraries\esp32cam-main 
Using library FS at version 1.0 in folder: C:\Users\Mohammad\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS 
exit status 1
Error compiling for board ESP32 Wrover Module.

yoursunny, 2022-07-24:
https://github.com/yoursunny/esp32cam/issues/23#issuecomment-1047664435

closed 2022-07-24 by yoursunny