yoursunny/esp32cam #37: 'camera_sensor_info_t' was not declared in this scope

rezwanMahadi, 2022-10-02:
_**What would be the problem?
@yoursunny**_

Arduino: 1.8.15 (Windows 10), Board: "ESP32 Wrover Module, Huge APP (3MB No OTA/1MB SPIFFS), QIO, 80MHz, 921600, None"

C:\Users\User\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp: In member function 'esp32cam::ResolutionList esp32cam::CameraClass::listResolutions() const':

C:\Users\User\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:3: error: 'camera_sensor_info_t' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);

   ^
C:\Users\User\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:25: error: 'info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                         ^
C:\Users\User\Documents\Arduino\libraries\esp32cam-main\src\esp32cam.cpp:30:70: error: 'esp_camera_sensor_get_info' was not declared in this scope
   camera_sensor_info_t* info = esp_camera_sensor_get_info(&sensor->id);
                                                                      ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Not used: D:\arduino-1.8.15-windows\arduino-1.8.15\libraries\WiFi

exit status 1

Error compiling for board ESP32 Wrover Module.

yoursunny, 2022-10-02:
See #36, which is exactly the same issue, and pinned on top of the issue list.

closed 2022-10-02 by yoursunny

rezwanMahadi, 2022-10-02:
Thanks, for your suggestion. It worked!