yoursunny/esp32cam #58: Cannot bind non-const lvalue reference of type 'Client&' to an rvalue of type 'Client'

victorbnl, 2024-06-26:
I have a class which has a `WebServer webServer` member, as well as a setup function, in which I declare the handlers, including one for testing the camera, as follows.

```cpp
  webServer.on("/photo", HTTP_GET, [&] {
    auto frame = esp32cam::capture();
    webServer.setContentLength(frame->size());
    webServer.send(200, "image/jpeg");
    frame->writeTo(webServer.client());
  });
  ```

However this produces the following error:

```
src/server.cpp: In lambda function:
src/server.cpp:37:36: error: cannot bind non-const lvalue reference of type 'Client&' to an rvalue of type 'Client'
     frame->writeTo(webServer.client());
                    ~~~~~~~~~~~~~~~~^~
In file included from .pio/libdeps/debug/esp32cam/src/internal/mjpeg.hpp:4,
                 from .pio/libdeps/debug/esp32cam/src/esp32cam.h:11,
                 from src/server.cpp:3:
.pio/libdeps/debug/esp32cam/src/internal/frame.hpp:49:8: note:   initializing argument 1 of 'bool esp32cam::Frame::writeTo(Client&, int)'
   bool writeTo(Client& os, int timeout = 10000);
        ^~~~~~~
*** [.pio/build/debug/src/server.cpp.o] Error 1
```

yoursunny, 2024-06-27:
You have arduino-esp32 v2.x or older, in which `WebServer::client()` returns a copy of `WiFiClient`:
https://github.com/espressif/arduino-esp32/blob/2.0.17/libraries/WebServer/src/WebServer.h#L112

esp32cam library has been upgraded to work with arduino-esp32 v3.x or newer, in which `WebServer::client()` returns a reference of `NetworkClient&`:
https://github.com/espressif/arduino-esp32/blob/3.0.0/libraries/WebServer/src/WebServer.h#L161

You need to upgrade to arduino-esp32 v3.x through Arduino Boards Manager.

victorbnl, 2024-06-27:
Uh, PlatformIO does not seem to support arduino-esp32 v3.x natively at the moment. Adding the following into my project’s platform.ini worked as a workaround.

```
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-rc1
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
```

closed 2024-06-27 by victorbnl