yoursunny/esp32cam #61: error: 'enable_if_t' in namespace 'std' does not name a template type

DinossauroBebado, 2025-02-02:
trying to compile the code in plataform io and arduino ide and both return the same error : 
```
In file included from .pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.cpp:1:
.pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.hpp:64:8: error: 'enable_if_t' in namespace 'std' does not name a template type
   std::enable_if_t<std::is_assignable_v<std::function<void(Settings&)>, Fn>, bool> //
        ^~~~~~~~~~~
.pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.hpp:64:3: note: suggested alternative: 'enable_if'
   std::enable_if_t<std::is_assignable_v<std::function<void(Settings&)>, Fn>, bool> //
   ^~~
   enable_if
In file included from .pio/libdeps/esp32cam/esp32cam/src/esp32cam/config.cpp:2:
.pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.hpp:64:8: error: 'enable_if_t' in namespace 'std' does not name a template type
   std::enable_if_t<std::is_assignable_v<std::function<void(Settings&)>, Fn>, bool> //
        ^~~~~~~~~~~
.pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.hpp:64:3: note: suggested alternative: 'enable_if'
   std::enable_if_t<std::is_assignable_v<std::function<void(Settings&)>, Fn>, bool> //
   ^~~
   enable_if
In file included from .pio/libdeps/esp32cam/esp32cam/src/esp32cam.h:11,
                 from src/main.cpp:3:
.pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.hpp:64:8: error: 'enable_if_t' in namespace 'std' does not name a template type
   std::enable_if_t<std::is_assignable_v<std::function<void(Settings&)>, Fn>, bool> //
        ^~~~~~~~~~~
.pio/libdeps/esp32cam/esp32cam/src/esp32cam/camera.hpp:64:3: note: suggested alternative: 'enable_if'
   std::enable_if_t<std::is_assignable_v<std::function<void(Settings&)>, Fn>, bool> //
   ^~~
   enable_if
*** [.pio/build/esp32cam/lib7e5/esp32cam/esp32cam/camera.cpp.o] Error 1
*** [.pio/build/esp32cam/src/main.cpp.o] Error 1
*** [.pio/build/esp32cam/lib7e5/esp32cam/esp32cam/config.cpp.o] Error 1
```

yoursunny, 2025-02-02:
`std::enable_if_t` is available since C++17.

Please verify that your have the latest arduino-esp32 installed.
[arduino-esp32 v3.1.1](https://github.com/espressif/arduino-esp32/releases/tag/3.1.1) comes with [esp-x32 toolchain version 2405](https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20240530) that includes gcc 13.2, in which the default C++ standard has been raised to `-std=c++17`.

DinossauroBebado, 2025-02-03:
Thank you adding `build_unflags = -std=gnu++11
build_flags = -std=gnu++2a` in the .ini file in the plataform io fixed my issue

closed 2025-02-03 by DinossauroBebado