yoursunny/esp32cam #64: error

dudu6-ai, 2025-06-15:
thread 'main' panicked at 'assertion failed: `(left != right)`
  left: `0`,
 right: `0`: Failed to get path name. Error code: 3', main.rs:65:9
stack backtrace:
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
exit status 101

Compilation error: exit status 101

yoursunny, 2025-06-15:
What did you do that caused the above error?

dudu6-ai, 2025-06-15:
Just following the video on the internet

yoursunny, 2025-06-15:
Where's the video?
Please post a link.

dudu6-ai, 2025-06-15:
esp32-cam教程：https://blog.csdn.net/qq_48778364/article/details/126451600  from CSDN

closed 2025-06-15 by yoursunny

dudu6-ai, 2025-06-15:
> This tutorial is not using my library and thus is irrelevant to this repository.

are  you have the code operation steps？

dudu6-ai, 2025-06-15:
what can I do？