yoursunny/esp32cam #65: Still JPEG image is broken in the AsyncCam example

gkostov, 2025-09-21:
Hi,

This issue is only for still images in the AsyncCam example. The WifiCam example serves the images (both JPEG and BMP) correctly. MJPEG is fine in both examples.

Board is Freenove ESP32 Wrover Cam clone
Environment is esp32@3.3.1

Still JPEG images with resolution above 320x240 (that is 320x320 and above) show broken like this

[image]

Once in a while the image shows correctly but fails maybe 9 out of 10 times. Occasionally it does not load at all - the Serial log shows _"[ 1453435] StillResponse(0x3ffdf634) created"_, the browser shows the request as Pending and nothing happens for minutes. Until the browser aborts the request and the Serial immediately prints _"[ 1703906] StillResponse(0x3ffdf634) deleted"_

It happens in both Chrome and Safari so it's unlikely to be browser related.
I tried changing the other settings in the example but images were still broken.

I wonder if this has occurred before and if there is a known fix?

Thanks

yoursunny, 2025-10-18:
I attempted several things, but I still cannot conclude whether the problem occurs during image capture or async transmission.

## convert to BMP

I inserted code in `StillResponse::_fillBuffer` to invoke `m_frame->toBmp()` upon retrieving the frame.
The resulting BMP image appears correctly.

## convert to RGB888 then re-encode as JPG

I modified `Image::toJpeg` function so that it supports JPEG->JPEG conversion.
This uses RGB888 format as an intermediary, because `fmt2jpg` does not support JPEG as an input format.

Then, I inserted code in `StillResponse::_fillBuffer` to invoke `m_frame->toJpeg(80)` upon retrieving the frame.
The resulting re-encoded JPEG image still appears broken.

## copy image to malloc'd buffer

I inserted code in `StillResponse::_fillBuffer` to copy the frame buffer to a newly allocated buffer.
The copied image buffer still appears broken.

yoursunny, 2025-10-18:
I have determined that the broken JPEG image is caused by async transmission.

1. I inserted code at the end of `StillResponse::_fillBuffer` to print how many octets were returned to AsyncWebServer and its SHA256 digest.
2. It's noticed that a JPEG image of 480x320 resolution is returned from one `_fillBuffer` call, while a JPEG image of 640x480 resolution needs two `_fillBuffer` calls.
3. After setting 640x480 resolution, I downloaded a JPEG image with `wget` command.
4. It's noted that the downloaded file has 8398 octets, while the serial log indicates that the image should have 8483 octets.
5. I then checked the SHA256 digest of the first 5760 octets and the last 2723 octets, which is the split shown in the serial logs.
6. It's noted that the first 5760 octets have the wrong digest, while the last 2723 octets have the correct digest.
7. This indicates that AsyncWebServer somehow lost some octets returned from the first `_fillBuffer` call.

```text
[  704158] StillResponse(0x3ffdd540) created
[  704600] StillResponse(0x3ffdd540) frame has 8483 octets
[  704601] StillResponse(0x3ffdd540) sent buflen=5760 len=5760 index=5760
[  704602] StillResponse(0x3ffdd540) digest=QXu3zDDA/SodEz6SsQk0PoB3cGNaBOIZbs87cc+3Wls=
[  704642] StillResponse(0x3ffdd540) sent buflen=4308 len=2723 index=8483
[  704643] StillResponse(0x3ffdd540) digest=N5Ta191ONm1HyGUYLXau5OUCwk1iuaByRi2qJs3Jc5U=
[  704654] StillResponse(0x3ffdd540) sent to client
[  704655] StillResponse(0x3ffdd540) deleted
```

```shell
sunny@sunnyB:~$ wget http://192.168.5.85/cam.jpg
--2025-10-17 21:54:34--  http://192.168.5.85/cam.jpg
Connecting to 192.168.5.85:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [image/jpeg]
Saving to: ‘cam.jpg’

cam.jpg                               [ <=>                                                       ]   8.20K  --.-KB/s    in 0.05s

2025-10-17 21:54:35 (178 KB/s) - ‘cam.jpg’ saved [8398]

sunny@sunnyB:~$ wc cam.jpg
 141  329 8398 cam.jpg

sunny@sunnyB:~$ head -c 5760 cam.jpg | sha256sum | awk '{print $1}' | xxd -r -p | base64
n4RoBN1cPrerdL/14PtNGMBAHe5dZOw/LyMT3+hUDuk=

sunny@sunnyB:~$ tail -c 2723 cam.jpg | sha256sum | awk '{print $1}' | xxd -r -p | base64
N5Ta191ONm1HyGUYLXau5OUCwk1iuaByRi2qJs3Jc5U=
```

gkostov, 2025-11-25:
This has been fixed with the latest ESPAsyncWebServer release. It's working well now.

Cheers